fossi-foundation/open-pdks #428: Trouble installing open_pdks

Blue-moon-pix, 2024-01-27:
Hi, I tried to install open_pdk in my ubuntu system. While running make command I got the following error.
[image]

Here is the log file
[sky130A_make.log](https://github.com/RTimothyEdwards/open_pdks/files/14071276/sky130A_make.log)
[sky130B_make.log](https://github.com/RTimothyEdwards/open_pdks/files/14071282/sky130B_make.log)

d-m-bailey, 2024-01-27:
@Blue-moon-pix don't know if this will make difference, but looks like your using an older version of magic.
```
Magic 8.3 revision 50 - Compiled on Friday 19 January 2024 04:15:11 PM IST.
```
The current version is `8.3.460`. Can you install the most recent version?

Blue-moon-pix, 2024-01-27:
I installed recent version of magic. when I ran the make command my terminal stuck like this. I tried 3 times but still same
[image]

d-m-bailey, 2024-01-27:
@Blue-moon-pix Do you need all the standard cell libraries?

volare is a quick way to download pre-built libraries if you don't need the extra `hs` or `hdll` etc libraries.

RTimothyEdwards, 2024-01-27:
@Blue-moon-pix :  I think likely the "stuck" state is still processing---Unless you have a pretty high-end, multi-core machine, the process of compiling the PDK can take a long time;  if you have low memory overhead and it causes a lot of swapping, it might be a _very_ long time.  Make sure you give it at _least_ an hour to finish, even if it's not paging.  Or, you can take Mitch's advice and use volare (https://github.com/efabless/volare), which installs pre-compiled PDKs.

Blue-moon-pix, 2024-01-27:
Got it. Thanks for the advice :)

Blue-moon-pix, 2024-01-28:
Can i install one library at a time?

RTimothyEdwards, 2024-01-28:
@Blue-moon-pix :  Yes, although it's not really set up to do that conveniently.  If you run `configure --help` you will get a list of all the libraries open_pdks knows about.  Some are enabled by default and some are disabled by default.  If you want to cut down the number of libraries installed, you will need to run `configure` with a bunch of disables for the libraries that you don't want, like `--disable-sc-ms-sky130` (note that there are libraries supported for both sky130 and gf180mcu;  only libraries for the PDK that has itself been enabled will be installed;  so if you use `--enable-sky130-pdk`, for example, then you don't need to worry about disabling all of the gf180 libraries because none of them is going to be installed anyway.

Blue-moon-pix, 2024-01-28:
> @Blue-moon-pix :  Yes, although it's not really set up to do that conveniently.  If you run `configure --help` you will get a list of all the libraries open_pdks knows about.  Some are enabled by [...]

I have one doubt though, if I run 'configure' with disabling low power and low speed and run 'make' to install it and again I run configure with disabling other libraries and enable low power and low speed then other libraries got deleted or its unchanged?

Ps: I have only 2 gigs of ram for my Ubuntu VM

RTimothyEdwards, 2024-01-28:
@Blue-moon-pix :  Each library is entirely self-contained, and if you generate libraries one at a time, `make install` will add them to whatever is already there and will not delete any libraries that were previously installed.

Blue-moon-pix, 2024-01-28:
> @Blue-moon-pix :  Each library is entirely self-contained, and if you generate libraries one at a time, `make install` will add them to whatever is already there and will not delete any libraries that were previously installed.

Alright will try that

closed 2024-01-28 by Blue-moon-pix