fossi-foundation/open-pdks #435: Sky130 tech LEF: Add (dummy) CUT layer after wells?

ubfx, 2024-03-07:
Hello everyone,

this topic was briefly discussed in the early phases of PDK release on the google repos. Since things have stabilized a bit now and open_pdks has pretty much taken over the role of the "golden" PDK while the google repos are dormant, I'd like to bring it up again here.

The issue is that with the addition of `pwell` and `nwell` layers to the tlef (quite a while back), they're not compliant with the LEF/DEF standard anymore because they're missing a CUT layer to the local interconnect. Although OpenROAD seems to handle this without problems, certain commercial tools don't. I understand that the focus of the PDK is to enable the Open Source tools (which is also what I'm working on), but in practice, it's still useful to be able to run the commercial tools for cross-checking. And judging by github issues on the old repos, there's certainly interest to get the PDK running on the commercial tools.
I also know that it's quite simple to fix locally either by following the advice in the old PR below, or just by adding a dummycut layer between `pwell` and `li1`, but the fragmentation of the PDK into 20 variants with slightly different patches doesn't help the cause.

So my question: Could we add poly+cut or just a dummy cut layer between `pwell` and `li1` in the master tlefs?

Original Issue: https://github.com/google/skywater-pdk/issues/308
PRs: https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/pull/4 https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/pull/3

ubfx, 2024-03-07:
> I think it needs both the poly masterslice definition as well as the licon cut definition to make the commercial tools happy. Apparently they also need the overlap definition?

I'm currently running Cadence successfully with just the "dummy cut" layer, but I don't know how it is with other vendors and I agree it would probably be best to fix it properly.

> I'm happy to make the change to the efabless fork of the PDK repositories, but a pull request needs to be made for the technology LEF files of _all_ of the standard cell libraries.

Thank you! I'll do a few more tests and then I'll start making those PRs.

ubfx, 2024-03-07:
So the LEF/DEF reference says :

> Defines masterslice (nonrouting) or overlap layers in the design. Masterslice layers are
> typically polysilicon layers and are only needed if the cell MACROs have pins on the polysilicon
> layer. 

Since our cells have no pins on poly, I don't think we need it. But we do need `licon`. Now the question is, do we need to specify any rules (`ENCLOSURE`, `SPACING`, ...) for `licon`? It seems like the PnR tool wouldn't need to know about any potential rules related to licon because it would never have to generate vias there, or am I missing something?

RTimothyEdwards, 2024-03-08:
@ubfx:  No, you've got it about right.  The issue here is that the tools have absolutely no reason to be messing with the licon layer.  The open source tools like OpenROAD and magic all understand that everything works perfectly well without it.  I do not know why the commercial tools (and, maybe, the spec?) insist upon it being defined.  There are other cases related to the LEF/DEF specs where (1) the spec was ambiguously worded;  (2) some commercial tool interpreted the spec in the dumbest possible way;  and (3) that interpretation became the de facto interpretation.  I feel like this is another such example.

ubfx, 2024-03-08:
I see what you mean. Indeed, I wasn't able to find where the LEF/DEF specification requires the cut layer. It just says that you have to define the layers from the bottom up (not that you can't skip a layer).

At least Cadence seems to be happy with this configuration:
```
[...]
  TYPE MASTERSLICE ;
  PROPERTY LEF58_TYPE "TYPE PWELL ;" ;
END pwell

LAYER licon
  TYPE CUT ;
END licon

LAYER li1
  TYPE ROUTING ;
  DIRECTION VERTICAL ;
[...]
```

which would be the "minimally invasive" way to fix this.

ubfx, 2024-03-09:
Thank you for merging all of these!

closed 2024-03-10 by ubfx