fossi-foundation/open-pdks #436: sky130_fd_sc_hd__macro_sparecell has instances with connections in the wrong order.

d-m-bailey, 2024-03-28:
At least `version 1.0.470` and before.

The instance connections in `sky130_fd_sc_hd__macro_sparecell` do not match the respective subcircuit port orders.

Here's the `sky130_fd_sc_hd__macro_sparecell` definition.
```
.subckt sky130_fd_sc_hd__macro_sparecell VGND VNB VPB VPWR LO
Xsky130_fd_sc_hd__nand2_2_1 sky130_fd_sc_hd__nor2_2_1/B LO LO VPB VNB VGND VPWR sky130_fd_sc_hd__nand2_2
Xsky130_fd_sc_hd__nand2_2_0 sky130_fd_sc_hd__nor2_2_0/A LO LO VPB VNB VGND VPWR sky130_fd_sc_hd__nand2_2
Xsky130_fd_sc_hd__inv_2_0 sky130_fd_sc_hd__inv_2_0/A sky130_fd_sc_hd__inv_2_0/Y VPB
+ VNB VPWR VGND sky130_fd_sc_hd__inv_2
Xsky130_fd_sc_hd__inv_2_1 sky130_fd_sc_hd__inv_2_1/A sky130_fd_sc_hd__inv_2_1/Y VPB
+ VNB VPWR VGND sky130_fd_sc_hd__inv_2
Xsky130_fd_sc_hd__nor2_2_0 sky130_fd_sc_hd__nor2_2_0/A sky130_fd_sc_hd__inv_2_0/A
+ sky130_fd_sc_hd__nor2_2_0/A VPB VNB VGND VPWR sky130_fd_sc_hd__nor2_2
Xsky130_fd_sc_hd__nor2_2_1 sky130_fd_sc_hd__nor2_2_1/B sky130_fd_sc_hd__inv_2_1/A
+ sky130_fd_sc_hd__nor2_2_1/B VPB VNB VGND VPWR sky130_fd_sc_hd__nor2_2
Xsky130_fd_sc_hd__conb_1_0 LO sky130_fd_sc_hd__conb_1_0/HI VPB VNB VGND VPWR sky130_fd_sc_hd__conb_1
.ends
```

As an example, here is the `sky130_fd_sc_hd__inv_2` definition
```
.subckt sky130_fd_sc_hd__inv_2 A VGND VNB VPB VPWR Y
X0 Y A VGND VNB sky130_fd_pr__nfet_01v8 w=650000u l=150000u
X1 VPWR A Y VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
X2 VGND A Y VNB sky130_fd_pr__nfet_01v8 w=650000u l=150000u
X3 Y A VPWR VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
.ends
```

The inverter subckt definition has the output as the last port, but the instances have `VGND` as the last port.

I believe this is the correct version
```
.subckt sky130_fd_sc_hd__macro_sparecell VGND VNB VPB VPWR LO
Xsky130_fd_sc_hd__nand2_2_1 LO LO VGND VNB VPB VPWR sky130_fd_sc_hd__nor2_2_1/B sky130_fd_sc_hd__nand2_2
Xsky130_fd_sc_hd__nand2_2_0 LO LO VGND VNB VPB VPWR sky130_fd_sc_hd__nor2_2_0/A sky130_fd_sc_hd__nand2_2
Xsky130_fd_sc_hd__inv_2_0 sky130_fd_sc_hd__inv_2_0/A VGND VNB VPB
+ VPWR sky130_fd_sc_hd__inv_2_0/Y sky130_fd_sc_hd__inv_2
Xsky130_fd_sc_hd__inv_2_1 sky130_fd_sc_hd__inv_2_1/A VGND VNB VPB
+ VPWR sky130_fd_sc_hd__inv_2_1/Y sky130_fd_sc_hd__inv_2
Xsky130_fd_sc_hd__nor2_2_0 sky130_fd_sc_hd__nor2_2_0/A
+ sky130_fd_sc_hd__nor2_2_0/A VGND VNB VPB VPWR sky130_fd_sc_hd__inv_2_0/A sky130_fd_sc_hd__nor2_2
Xsky130_fd_sc_hd__nor2_2_1 sky130_fd_sc_hd__nor2_2_1/B
+ sky130_fd_sc_hd__nor2_2_1/B VGND VNB VPB VPWR sky130_fd_sc_hd__inv_2_1/A sky130_fd_sc_hd__nor2_2
Xsky130_fd_sc_hd__conb_1_0 VGND VNB VPB VPWR sky130_fd_sc_hd__conb_1_0/HI LO sky130_fd_sc_hd__conb_1
.ends
```

This is the only hierarchical cell in the library and thus the only cell affected by reordering the pins on the standard cells.

RTimothyEdwards, 2024-03-28:
Sounds right but needs to be an issue in https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd and needs to be fixed in that repository.

RTimothyEdwards, 2024-03-28:
It's quite possible that the pin order *was* changed by open_pdks, but got missed in the conversion where I pulled all the hack processing out of open_pdks as I fixed issues in the PDK repository itself.  But going forward, fixes are supposed to be done in the PDK;  open_pdks hacks (or patches, if you will) should only be used as a fallback in case the PDK source repository cannot be modified.

d-m-bailey, 2024-03-28:
Moved to the pdk repo

closed 2024-03-28 by d-m-bailey