fossi-foundation/open-pdks #440: `sky130_fd_sc_hd` corners `ff_n40C_1v95` and `ss_n40C_1v60` .lib refer to missing lookup tables

passant5, 2024-05-07:
Both of them refer to `ccsn_*` lookup tables that are not available in the .lib. Maybe they got copied from the ccsnoise version of the same corner by mistake?

RTimothyEdwards, 2024-05-07:
Seems to be a pretty widespread problem, too.  Doing
```
grep ccsn sky130_fd_sc_*/timing/* | grep -v ccsnoise | cut -d' ' -f1 | uniq | cut -d. -f1-2 | uniq
```
yields
```
sky130_fd_sc_hd/timing/sky130_fd_sc_hd__ff_n40C_1v95.lib
sky130_fd_sc_hd/timing/sky130_fd_sc_hd__ss_n40C_1v60.lib
sky130_fd_sc_hdll/timing/sky130_fd_sc_hdll__ff_n40C_1v95.lib
sky130_fd_sc_hdll/timing/sky130_fd_sc_hdll__ss_n40C_1v60.lib
sky130_fd_sc_hs/timing/sky130_fd_sc_hs__ff_n40C_1v95.lib
sky130_fd_sc_hs/timing/sky130_fd_sc_hs__ss_n40C_1v60.lib
sky130_fd_sc_hs/timing/sky130_fd_sc_hs__tt_025C_1v80.lib
sky130_fd_sc_hvl/timing/sky130_fd_sc_hvl__ff_n40C_5v50.lib:
sky130_fd_sc_hvl/timing/sky130_fd_sc_hvl__ff_n40C_5v50_lv1v95.lib:
sky130_fd_sc_hvl/timing/sky130_fd_sc_hvl__ss_n40C_1v65.lib:
sky130_fd_sc_hvl/timing/sky130_fd_sc_hvl__ss_n40C_1v65_lv1v60.lib:
sky130_fd_sc_ls/timing/sky130_fd_sc_ls__ff_n40C_1v95.lib
sky130_fd_sc_ls/timing/sky130_fd_sc_ls__ss_n40C_1v60.lib
sky130_fd_sc_ls/timing/sky130_fd_sc_ls__tt_025C_1v80.lib
sky130_fd_sc_ms/timing/sky130_fd_sc_ms__ff_n40C_1v95.lib
sky130_fd_sc_ms/timing/sky130_fd_sc_ms__ss_n40C_1v60.lib
sky130_fd_sc_ms/timing/sky130_fd_sc_ms__tt_025C_1v80.lib
```

RTimothyEdwards, 2024-05-07:
It looks pretty simple to solve.  When generating the non-ccsnoise versions of the files with ccsnoise data, only the tables got removed, and none of the contents.  It's pretty easy to write a script that removes them.  Looking at the original SkyWater files, the only difference between the two is that the ccsnoise version of the files has extra entries for the ccsnoise tables.  In all other respects the two files are the same.

passant5, 2024-05-08:
@RTimothyEdwards Thanks

closed 2024-05-08 by passant5