fossi-foundation/open-pdks #445: Different random variables for each device in monte-carlo simulation

rnunes2311, 2024-05-21:
Hi,
It looks like every device is getting independent/different random variables assigned in a monte-carlo simulation.
I also realised that the resistors were not getting any variation. This appears to be a mistake in the parameters_res_nom.spice. For example:
```.param sw_sky130_fd_pr__res_high_po_rs = {325.0+corner_factor*0.0} + process_mc_factor*MC_PR_SWITCH*GAUSS(0,0.035,1)```

GAUSS generates random values normalised to the mean, which is 0. Therefore it always generates 0, I believe.

RTimothyEdwards, 2024-05-22:
I'm not getting that.  Gaussian random variables calculated in a `.param` statement should be calculated once, not for every device.  And for the question above, `process_mc_factor` is 1 (which I think corresponds to 3 sigma variation), and`MC_PR_SWITCH` is 1 when the monte carlo corner is selected, so the parameter calculated should be (325 + (random number with zero mean)).

rnunes2311, 2024-05-22:
For the resistors I see in simulation no variation at all. The ngspice manual states that the `gauss` function generates random numbers relative to the mean. For that parameter it is generating 325 + (random number with 0 mean and 0 sigma) = 325. For most gaussian random variables in the pdk the `agauss` function is used, which generates random numbers with an absolute sigma value. I see no variation at all in the resistors with a monte-carlo simulation. By modifying the expression I could get the 3.5 % sigma variation.
I also see mismatch in transistors, which should not happen and when I place resistors in parallel and series to increase the area while still keeping the same equivalent resistance value the equivalent resistance variation is not 3.5 % 1 sigma, because every resistor is getting a different value assigned.