fossi-foundation/open-pdks #448: Unable to install open_pdk

emmagood987, 2024-06-19:
Hi,

I am trying to install OpenPDK in my WSL2 based Ubuntu 24.04. I am getting error while doing the
process and unable to proceed. I hope someone can help me.

First, I installed Xschem, Magic and ngspice successfully. Then I started installing OpenPDK.
I ran the following commands:

1. git clone https://github.com/RTimothyEdwards/open_pdks
This command ran OK.

2. cd open_pdks

3. ./configure --enable-sky130-pdk --prefix=/home/emma
This command also went OK. Pls see screenshot_1.jpg

[image]

4. make
After 45 min this command exited with ‘error2’. The log has several of the following errors:
……..
……..
make[3]: *** [Makefile:1524: primitive-B] Error 1
make[3]: Leaving directory '/home/nir/open_pdks/sky130'
make[2]: *** [Makefile:1407: primitive-build-B] Error 2
make[2]: Leaving directory '/home/nir/open_pdks/sky130'
make[1]: *** [Makefile:1053: all-B] Error 2
make[1]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:1524: primitive-A] Error 1
make[3]: Leaving directory '/home/nir/open_pdks/sky130'
make[2]: *** [Makefile:1407: primitive-build-A] Error 2
make[2]: Leaving directory '/home/nir/open_pdks/sky130'
make[1]: *** [Makefile:1053: all-A] Error 2
make[1]: Leaving directory '/home/nir/open_pdks/sky130'
make: *** [Makefile:132: tech-sky130] Error 2

Pls see the screenshot_2.jpg

[image]

5. sudo make install
This command is exiting with ‘error2’. Pls see the screenshot_3.jpg

[image]

Can you pls tell me why I am getting this error and most importantly, how to overcome it?

I look forward to your response.

Thanks,
Emma.

d-m-bailey, 2024-06-19:
@emmagood987 Thanks for the detailed logs. Do you need to install open_pdks from source or can you use the precompiled versions available with [volare here](https://github.com/efabless/volare)?

emmagood987, 2024-06-19:
> @emmagood987 Thanks for the detailed logs. Do you need to install open_pdks from source or can you use the precompiled versions available with [volare here](https://github.com/efabless/volare)?

Thanks for your reply. I can use volare. But I dont know how to. Where can I find the detailed steps on how to install through volare pls? Or if you can type them out sequentially, it would be just great.

d-m-bailey, 2024-06-19:
The installation steps are [detailed here](https://github.com/efabless/volare/blob/main/Readme.md).

```
export PDK_ROOT=<pdk directory>
export PDK=<process>  # sky130A, sky130B, etc.
python3 -m pip install --upgrade --no-cache-dir volare
volare ls-remote
volare enable <desired-commit>
```

emmagood987, 2024-06-19:
> The installation steps are [detailed here](https://github.com/efabless/volare/blob/main/Readme.md).
.
> python3 -m pip install --upgrade --no-cache-dir volare

Ok. I have tried the steps at the Volare git page. However the above command is not working at my end. I have already opened an issue at the Volare git page. You can see the issue at the link [here](https://github.com/efabless/volare/issues/91). Do let me know if you have the solution for it.

Emma.