fossi-foundation/open-pdks #449: NGSpice files missing in Open-PDK install?

MOS5500, 2024-07-24:
Hi to all,
I have installed open_Pdk and it make several directories in /usr/local/share/pdk/sky130A/libs.tech/ (after a strange delay for configuring (??) files for magic)

All directories contains files, I can use correctly Xschem and magic.
But if I launch a simulation from Xschem, I obtain an error because the sub-directory ngspice is empty.

Why?

RTimothyEdwards, 2024-07-24:
Good question.
The first thing is to pin down where the issue occurs.  The upstream source, if you let open_pdks do the work of cloning all the repositories, should end up in the directory (in open_pdks) sources/sky130_fd_pr/.  If that exists (and is populated), then what about the staging directory (in open_pdks) sky130/sky130A/libs.tech/ngspice (also sky130/sky130A/libs.tech/combined)?

MOS5500, 2024-07-24:
> The upstream source, if you let open_pdks do the work of cloning all the repositories, should end up in the directory (in open_pdks) sources/sky130_fd_pr/

In this directory there is only a subdir ".git"

> If that exists (and is populated), then what about the staging directory (in open_pdks) sky130/sky130A/libs.tech/ngspice (also sky130/sky130A/libs.tech/combined)?

This two dirs are completely empty.

At this point I imagine it's a repository cloning problem, but I had already tried to download it (and reinstall it) a couple of times.

MOS5500, 2024-07-25:
Is it possible to install manually the files?

RTimothyEdwards, 2024-07-25:
Yes, of course.  The only thing the Makefile is doing is "git clone" on a bunch of repositories.

I have heard of people having issues with timeouts of github, depending on where you are located in the world.  In some cases it can require multiple attempts, and/or running it at off-peak hours.  Working around that problem may require increasing the values for RETRIES_NO and RETRY_DELAY in scripts/download.sh.

But also, I see from the Makefile that if the directory is created but the clone does not complete, then the fact that the directory exists prevents the make target from re-running.  Assuming that all directories got created but some or all may not have completed the "git clone", then I suggest doing:
```
make update
```
This will do "git pull" on each of the repositories, and so will complete the clone operation if it failed the first time.

RTimothyEdwards, 2024-07-25:
Also, assuming that in all those directories at least the `.git/config` file was created, then you can just `cd` to any directory in `sources/` that is missing contents and run `git pull` as many times as you need to until it works.

MOS5500, 2024-07-30:
I finally did it!
After changing parameters, setting buffer sizes, global variables, changing dozens of git and OS parameters, 
I did this:
1) First I cloned the entire repository (and here the ngspice definitions are missing, which for obscure reasons, still does not clone).
2) I cloned the NGspice definitions repository separately
3) I copied the folder with ng definitions under open_pdks/sources, renaming it
4) I gave make

and finally now ngpice seems to work! :)

Time taken: almost 2 weeks.

OT
Isn't there an alternative system to GIT to do this?
It was a bloodbath, I had to fight for two weeks, changing parameters, dirtying the OS and probably without even remembering what I changed. Absurd.
Couldn't we make a damn interactive installer that does everything by itself?
Having a graphical installer like in Windows would be the best.
Because it is humanly impossible to extrapolate from all the make messages what the errors are and go and correct them, it seems to me an archaic and senseless system.
Losing two weeks, without having done anything productive is inconceivable. But even losing a single day to configure something that should be able to let you work is senseless.
This is NOT a criticism of the author of the package (to whom we thank for putting it all together), but of a system with many absurd limits and that really drives you crazy.
/OT