fossi-foundation/open-pdks #453: Git Fetch Fails with RPC Error and Invalid Index-Pack Output

gowreddysuresh, 2024-08-28:
Expected behavior: The fetch operation should complete successfully without errors.

Git version: git version 2.43.0

Repository URL: https://github.com/RTimothyEdwards/open_pdks

Operating System: Ubuntu 20.04 LTS

Error:
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 4471 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 28 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

gowreddysuresh, 2024-08-28:
@proppy  please solve this error

RTimothyEdwards, 2024-08-29:
@gowreddysuresh :  This is not a problem that is correctable in the PDK.  You are experiencing a timeout with the github servers.

You may wish to use volare (https://github.com/efabless/volare) to install a pre-built PDK instead of trying to download and build and install the PDK from source material.

gowreddysuresh, 2024-08-29:
@RTimothyEdwards  sky130 pdk was installed successfully using volare but when it is coming to install gf180mcu it is showing this error 

(myenv) vlsi@vlsi:~/gits$ volare enable --pdk gf180mcu 0fe599b2afb6708d281543108caf8310912f54af
Version 0fe599b2afb6708d281543108caf8310912f54af not found locally, attempting 
to download…
Downloading common.tar.zst… ━━━━━━━━━━━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━  40% 0:06:10
The read operation timed out
please solve this  error

gowreddysuresh, 2024-09-02:
@proppy please give an update on this issue

proppy, 2024-09-02:
@gowreddysuresh: as @RTimothyEdwards commented, this is likely due to communication issue between your computer and github server (this applies to downloading the PDK sources as well as using volare, because it relies on github for artifact storage).

I would suggest to attempt to run the same commands on a cloud server environment (aws, azure or gcp) to verify this hypothesis.

@RTimothyEdwards I think this can be closed as it's not really something that can be "fixed" in `open_pdks`.

Tallboy420, 2024-09-09:
can i get an update because i am having the same problem as @gowreddysuresh

closed 2024-09-13 by RTimothyEdwards