fossi-foundation/open-pdks #455: Seeking assistance for make command failure in Sky130 PDK

teodora1506, 2024-09-24:
Hi everyone,

I'm experiencing an issue while trying to clone the Openpdk repository on my Ubuntu machine. Here are the details:
Operating System: Ubuntu (virtual machine on Windows using Oracle VM VirtualBox)
Git Version: 2.34.1

Problem: I'm facing an issue when trying to run the make command,I encounter the following error/message: 
make[1]: Entering directory '/home/vboxuser/Desktop/chip_design/open_pdks/sky130'
if test "x" != "x"; then \
	FD_PR_COMMIT=`cat  | grep sky130_fd_pr | grep -v COMMIT | cut -d'"' -f4` ; \
fi ; \
if test "x../sources/sky130_fd_pr" != "x" ; then \
	if test -d "../sources/sky130_fd_pr" ; then \
		echo "Using existing installation of primitive library from ../sources/sky130_fd_pr" ; \
	else \
		echo "Downloading primitive library from ../sources/sky130_fd_pr" ; \
		../scripts/download.sh https://github.com/efabless/skywater-pdk-libs-sky130_fd_pr ../sources/sky130_fd_pr ${FD_PR_COMMIT} ; \
	fi ; \
fi
if test "x" != "x"; then \
	FD_IO_COMMIT=`cat  | grep sky130_fd_io | grep -v COMMIT | cut -d'"' -f4` ; \
fi ; \
if test "x../sources/sky130_fd_io" != "x" ; then \
	if test -d "../sources/sky130_fd_io" ; then \
		echo "Using existing installation of I/O library from ../sources/sky130_fd_io" ; \
	else \
		echo "Downloading I/O library from ../sources/sky130_fd_io" ; \
		../scripts/download.sh https://github.com/efabless/skywater-pdk-libs-sky130_fd_io ../sources/sky130_fd_io ${FD_IO_COMMIT} ; \
	fi ; \
fi
Downloading primitive library from ../sources/sky130_fd_pr
if test "x" != "x"; then \
	FD_SC_HD_COMMIT=`cat  | grep sky130_fd_sc_hd | grep -v COMMIT | cut -d'"' -f4` ; \
fi ; \
if test "x../sources/sky130_fd_sc_hd" != "x" ; then \
	if test -d "../sources/sky130_fd_sc_hd" ; then \
		echo "Using existing installation of high-density standard cell library from ../sources/sky130_fd_sc_hd" ; \
	else \
		echo "Downloading high-density standard cell library from ../sources/sky130_fd_sc_hd" ; \
		../scripts/download.sh https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hd ../sources/sky130_fd_sc_hd ${FD_SC_HD_COMMIT} ; \
	fi ; \
fi
Downloading I/O library from ../sources/sky130_fd_io
if test "x" != "x"; then \
	FD_SC_HDLL_COMMIT=`cat  | grep sky130_fd_sc_hdll | grep -v COMMIT | cut -d'"' -f4` ; \
fi ; \
if test "x../sources/sky130_fd_sc_hdll" != "x" ; then \
	if test -d "../sources/sky130_fd_sc_hdll" ; then \
		echo "Using existing installation of high-density low-leakage standard cell library from ../sources/sky130_fd_sc_hdll" ; \
	else \
		echo "Downloading high-density low-leakage standard cell library from ../sources/sky130_fd_sc_hdll" ; \
		../scripts/download.sh https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hdll ../sources/sky130_fd_sc_hdll ${FD_SC_HDLL_COMMIT} ; \
	fi ; \
fi
Downloading high-density standard cell library from ../sources/sky130_fd_sc_hd
Cloning https://github.com/efabless/skywater-pdk-libs-sky130_fd_pr to ../sources/sky130_fd_pr 
if test "x" != "x"; then \
	FD_SC_HVL_COMMIT=`cat  | grep sky130_fd_sc_hvl | grep -v COMMIT | cut -d'"' -f4` ; \
fi ; \
if test "x../sources/sky130_fd_sc_hvl" != "x" ; then \
	if test -d "../sources/sky130_fd_sc_hvl" ; then \
		echo "Using existing installation of high-voltage standard cell library from ../sources/sky130_fd_sc_hvl" ; \
	else \
		echo "Downloading high-voltage standard cell library from ../sources/sky130_fd_sc_hvl" ; \
		../scripts/download.sh https://github.com/efabless/skywater-pdk-libs-sky130_fd_sc_hvl ../sources/sky130_fd_sc_hvl ${FD_SC_HVL_COMMIT} ; \
	fi ; \
fi
Downloading high-density low-leakage standard cell library from ../sources/sky130_fd_sc_hdll
Cloning https://github.com/efabless/skywater-pdk-libs-sky130_fd_io to ../sources/sky130_fd_io 
if test "x" != "x"; then \
[... 231 lines cut ...]
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOFs:  39% (278/711), 20.36 MiB | 1.44 MiB/sB/s
fatal: fetch-pack: invalid index-pack output
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 3067 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 4262 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOFs:  99% (1643/1656), 20.79 MiB | 1.88 MiB/s
fatal: fetch-pack: invalid index-pack output
Cloning into '../sources/sky130_fd_io'... MiB | 733.00 KiB/s
Cloning into '../sources/sky130_fd_sc_hvl'... | 711.00 KiB/s
Cloning into '../sources/klayout_sky130'...
remote: Enumerating objects: 711, done.
remote: Counting objects: 100% (711/711), done.
remote: Enumerating objects: 7373, done.
remote: Enumerating objects: 1656, done.
remote: Counting objects: 100% (1656/1656), done.
remote: Counting objects: 100% (7373/7373), done.
remote: Compressing objects: 100% (1703/1703), done.

I would appreciate any help or guidance on how to resolve this issue. Thank you!

d-m-bailey, 2024-09-24:
@teodora1506 Looks like your github transfer timed out. You could try again to see if you get the same error.

If you're ok with a prebuilt pdk, I strongly recommend using volare.
```
git clone https://github.com/efabless/caravel_user_project
cd caravel_user_project
export PDK_ROOT=<where you want the pdk to be installed>
make pdk-with-volare
```

teodora1506, 2024-09-24:
I tried many times today and every time was the same.
Now I get this :
vboxuser@vboxuser-VirtualBox:~/Desktop/chip_design$ git clone https://github.com/efabless/caravel_user_project
Cloning into 'caravel_user_project'...
remote: Enumerating objects: 6222, done.
remote: Counting objects: 100% (41/41), done.
remote: Compressing objects: 100% (28/28), done.
error: RPC failed; curl 16 Error in the HTTP2 framing layerB/s
error: 4681 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

RTimothyEdwards, 2024-09-24:
@teodora1506 :  This is a very common problem with github mirror sites, especially from places like India and China.  You might just want to try at different times of day to see if you can manage to find a time when the server traffic is low.

However, `caravel_user_project` has a lot of history, so cloning the entire repository will require a very large download.  You may want to use `git clone https://github.com/efabless/caravel_user_project --depth=1` to avoid all the history, which will significantly cut down on the amount of data being transferred.

teodora1506, 2024-09-27:
I achieved to install sky130,but now i do not have  model file(tt,sf,ss etc), I do not know how to add them.