fossi-foundation/open-pdks #457: MacOS Klayout infinite window loop crashes computer

Revenantx86, 2024-10-10:
Hello, 

I have recently installed klayout and open pdk to my Mac. When I add the library to klayout on Mac. the python script launches infinite amount of klayout windows as like this: 

[image]

Do you know how I can fix this ? I suspect it's because of circular import loop in the script ? 

Refik

RTimothyEdwards, 2024-10-10:
Leo Moser (@mole99) made a pull request to the upstream library that will likely solve this problem:  https://github.com/efabless/sky130_klayout_pdk/pull/25.  The pull request is under review.

mole99, 2024-10-11:
Hi @Revenantx86, as Tim has said, there is now a new method to install the sky130 KLayout PDK from Efabless.
You should be able to install the package directly from KLayout's package manager:

[image]

Let me know how that goes for you.