fossi-foundation/open-pdks #459: sky130_tech/tech directory missing only tech

wspitts2, 2024-10-13:
The sky130_tech directory doesn't exist. only the Tech directory.

wspitts2, 2024-10-13:
./configure --enable-sky130-pdk --prefix=$(PREFIX) 
make 
cp: cannot stat '../sources/klayout_sky130/sky130_tech/tech/sky130/lvs/*': No such file or directory

HenryAEvans, 2024-10-14:
If you pass the `--with-reference` flag to the configure script, it will pull an older version of the klayout PDK repository, and this worked for me. Also I believe once #458 is merged this should be resolved.

RTimothyEdwards, 2024-10-14:
@wspitts2 , @HenryAEvans :  I just merged #458 and am testing now.

RTimothyEdwards, 2024-10-14:
Test checks out okay.  If klayout has been set up with the PDK as it was previously, then it requires going to Tools-->Manage Technologies, selecting sky130-->General, and in "Layer properties", changing "sky130.lyp" to "sky130A.lyp" (or set the base path and layer properties to "sky130B" for ReRAM support).