fossi-foundation/open-pdks #463: Magic doesn't list layers `scnmos` and `scpmos` used in e.g. `skywater-pdk-libs-sky130_fd_sc_hd`

fedeinthemix, 2024-10-25:
I run a simple `openlane2` trial run. When the flow finished, I checked the final gds file using
```bash
openlane --last-run --flow openinmagic designs/dff/config.json
```

The flow instantiates an instance of `sky130_fd_sc_hd__dfxtp_2`. I was unable to hide the nfet active area by right clicking in the layer list. I then queried the layers and magic reports that it uses the layer `scnmos`, but it's not in the layers list. Similarly for pfets.

I'm using `open_pdk` version 1.0.495 and magic 8.3.465.

RTimothyEdwards, 2024-10-25:
Certain layers are marked as not being user design layers.  The device types beginning with "sc" may only exist within standard cells.  They violate normal user design rules, but the standard cell layouts with these rule violations have been approved by SkyWater.  So the devices get special names when read into magic so that magic can handle the DRC rule exceptions.

Click on the menu option "Options-->Toolbar Hide Locked" (which is enabled by default) to show all of the layers, including the normally hidden ones.

Alternatively, it is often easier to use the command line to show or hide layers _en masse_:  e.g., `see no *; see alldiff,allpoly,nwell` or `see no allm5,allm4,allm3`.

fedeinthemix, 2024-10-25:
Alright, thanks for the explanation!

closed 2024-10-25 by fedeinthemix