fossi-foundation/open-pdks #464: Device Width and Fingers in mos generator seems incorrect.

christoph-weiser, 2024-11-08:
## Problem
The way magic generates mos devices is not respecting the Fingers field. It should divide the total Width by Fingers to give the correct width of each individual finger. 
The total width of the device should still be what was typed in the Width field.

## Implications

The problem is that this bug renders magics SPICE import function unusable without modifying the netlist manually before importing.

## Example

The input to the generator:
[image]

This is the layout generated: 
[image]

christoph-weiser, 2024-11-08:
Ahh ok, i assumed the SPICE netlist would be directly plugged into the device generator without any preprocessing. 

That makes sense then, but I think it would be good to indicate the Width means width per finger to avoid confusion, as it differs from what one would typically write in the schematic.

RTimothyEdwards, 2024-11-08:
Agreed.  Feel free to leave the issue open until I get around to implementing it.