fossi-foundation/open-pdks #465: Request for Xschem support for SKY130 special_nfet_01v8.sch

sifferman, 2024-11-09:
Hello. I am trying to use Xschem to create a circuit that uses the [special_nfet_01v8](https://github.com/efabless/skywater-pdk-libs-sky130_fd_pr/tree/main/cells/special_nfet_01v8) cell, but I noticed that open_pdks doesn't provide a `"sky130A/libs.tech/xschem/sky130_fd_pr/special_nfet_01v8.sch"` file out of the box.

I am happy to do a PR myself if I could get some pointers. I assume the required change is in [sky130/xcircuit/sky130_fd_pr.lps](https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/xcircuit/sky130_fd_pr.lps), but I haven't had the time to actually look through the file yet.

Thanks!

RTimothyEdwards, 2024-11-09:
"Special" devices are ones that do not meet DRC requirements and are only approved for use in specific cells that have been tested and characterized by SkyWater.  The "special_nfet_01v8" is used in a handful of standard cells and has a width of 0.36um or 0.39um, which is below the nFET minimum width requirement of 0.42um.  The devices should not be used outside of the standard cell layouts where they are found.

StefanSchippers, 2024-11-10:
@sifferman, as @RTimothyEdwards said you can just use a `nfet_01v8` transistor, set the W and L geometry to match the allowed values (these special devices have a fixed geometry, check the `*.pm3.spice` file) and change the model name.

Adding symbols for these 'special' devices will probably induce users to use them anywhere so I don't think it is a good move. @RTimothyEdwards do you agree?

RTimothyEdwards, 2024-11-10:
@StefanSchippers :  Yes, I agree.

sifferman, 2024-11-11:
Thanks for the help! My circuit now passes netgen lvs.

For future reference, this is the .sch snippet that worked for me: <https://github.com/sifferman/sky130_schematics/blob/1e51032f/schematics/dfxtp/sky130_fd_sc_hd__dfxtp_1.sch#L119-L124>

I will now close this issue.

closed 2024-11-11 by sifferman