fossi-foundation/open-pdks #466: Issue while running make command (Installing Open_Pdk)

neo-one-8, 2024-11-15:
<pre>ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
Traceback (most recent call last):
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 255, in &lt;module&gt;
    from setuptools.distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
make[3]: *** [Makefile:1873: alpha-B] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1487: alpha-build-B] Error 2
Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 255, in &lt;module&gt;
    from setuptools.distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 255, in &lt;module&gt;
    from setuptools.distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
make[3]: *** [Makefile:1551: io-A] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1423: io-build-A] Error 2
make[2]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 255, in &lt;module&gt;
    from setuptools.distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 255, in &lt;module&gt;
    from setuptools.distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
[... 93 lines cut ...]
ModuleNotFoundError: No module named &apos;setuptools.distutils&apos;

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File &quot;/home/mrvenkat/open_pdks/sky130/../common/foundry_install.py&quot;, line 257, in &lt;module&gt;
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named &apos;distutils&apos;
make[3]: *** [Makefile:1656: digital-hdll-A] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1447: digital-hdll-build-A] Error 2
make[3]: *** [Makefile:1873: alpha-A] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1487: alpha-build-A] Error 2
make[3]: *** [Makefile:1530: primitive-A] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1415: primitive-build-A] Error 2
make[3]: *** [Makefile:1803: digital-ms-A] Error 1
make[3]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[2]: *** [Makefile:1471: digital-ms-build-A] Error 2
make[2]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make[1]: *** [Makefile:1053: all-A] Error 2
make[1]: Leaving directory &apos;/home/mrvenkat/open_pdks/sky130&apos;
make: *** [Makefile:132: tech-sky130] Error 2
</pre>

neo-one-8, 2024-11-24:
Hi, Im still facing this issue. 
I looked at previous similar issues faced by others, which was dealt by either reinstalling the latest version of MAGIC or Open_Pdk. The versions of MAGIC and Open_Pdk that I have installed seems to be right, but unable to figure out whats wrong.
Any help would appreciated. Thank You.

matheus-VBLabs, 2024-12-01:
Hi Neo,

I encountered the same issue and discovered it was because I hadn't installed Magic. Once I installed Magic first and then the PDK, the error disappeared.

I noticed that you already have Magic installed, but this sequence worked for me: install Magic first, then the PDK.

I hope this helps!

closed 2024-12-18 by neo-one-8