fossi-foundation/open-pdks #475: The sky130_ef_fd_sc spice file for decap_12 is not pin compatible with the sky130_fd_sc_hd version.

d-m-bailey, 2025-02-18:
Version 1.0.493 and later

From `libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__decap_12.spice`
```
.subckt sky130_ef_sc_hd__decap_12 VGND VPWR VPB VNB
```

From `libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice`
```
.subckt sky130_fd_sc_hd__decap_12 VGND VNB VPB VPWR
```

The xschem symbol for `decap_12` has the pins in the `sky130_fd_sc_hd` order. This means that when using the symbol in a schematic, switching between the 2 libraries is not as simple as changing the prefix from `sky130_fd_sc_hd` to `sky130_ef_sc_hd`.

Changing the pin order in the `sky130_ef_sc_hd__decap_12.spice` file may be sufficient, but since this file was generated from an extracted layout, changing the pin order in the original mag file might be a good idea too.

The spice file is also referenced from verilog during device level LVS, but since the verilog to spice calls are by port name and not position, changing the port order in the spice subcircuit definition shouldn't cause a problem.

RTimothyEdwards, 2025-02-19:
@d-m-bailey :  There is no `.mag` source for these.  The sources are kept in `open_pdks` directory `sky130/custom/sky130_fd_sc_hd/` where there is only GDS, LEF, and verilog.  Since LEF and verilog do not have a definitive port order, the only ordering will come from the order in which text is read from GDS.  The only solution here is probably to copy the SPICE files for these cells back into the `custom/` directory, then add a `-spice` line for the installation of custom sources in the `digital-hd-%` recipe in the sky130 `Makefile` (line 1599).

d-m-bailey, 2025-02-19:
@RTimothyEdwards Thanks for the review. Wasn't there a push to have the default pin order extracted in a deterministically (possibly alphabetically)?

RTimothyEdwards, 2025-02-19:
Yes, that was Tim Ansell's idea, but it is not easy to do from within magic.  It could be done in python during the PDK build, but I think in this case it's easier to just provide the netlist.