fossi-foundation/open-pdks #480: .include corners/tt.spice brakes ngspice simulation

OtacilioNeto, 2025-05-15:
When .include corners/tt.spice is activated in 

`sky130A/libs.tech/combined/sky130.lib.spice`

the simulation using ngspice of this netlist 

```
** sch_path: /home/ota/workspace/IFPB_ENGE_CIA/Praticas e Projetos/01 Transistor/testeTransistor.sch
**.subckt testeTransistor
V1 net1 GND 0
V2 net2 GND 0.9
XM1 net2 net1 GND GND sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad=0.29 as=0.29 pd=2.58 ps=2.58 nrd=0.29 nrs=0.29 sa=0 sb=0 sd=0
+ mult=1 m=1
**** begin user architecture code

.control
	dc V1 0 3 0.01
	plot -i(V2)
.endc

.save all

** opencircuitdesign pdks install
.lib /home/ota/EDA/PDKs/sky130B/libs.tech/combined/sky130.lib.spice tt

**** end user architecture code
**.ends
.GLOBAL GND
.end

```
brakes whith this error message
```

** Copyright 2001-2024, The ngspice team.
** Please get your ngspice manual from https://ngspice.sourceforge.io/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Fri May  9 14:09:13 UTC 2025
******

Note: Compatibility modes selected: hs a

Warning: m=xx on .subckt line will override multiplier m hierarchy!

Warning: redefinition of .subckt sky130_fd_pr__nfet_01v8, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_01v8_lvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_01v8, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_03v3_nvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_05v0_nvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_01v8_lvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_01v8_hvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__esd_pfet_g5v0d10v5, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_g5v0d10v5, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_g5v0d10v5, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_g5v0d16v0, ignored
Warning: redefinition of .subckt sky130_fd_pr__esd_nfet_g5v0d10v5, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_20v0, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_20v0_iso, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_20v0_zvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_20v0, ignored
Warning: redefinition of .subckt sky130_fd_pr__nfet_20v0_nvt, ignored
Warning: redefinition of .subckt sky130_fd_pr__pnp_05v5_w0p68l0p68, ignored
Warning: redefinition of .subckt sky130_fd_pr__pnp_05v5_w3p40l3p40, ignored
Warning: redefinition of .subckt sky130_fd_pr__npn_05v5_w1p00l1p00, ignored
Warning: redefinition of .subckt sky130_fd_pr__npn_05v5_w1p00l2p00, ignored
Warning: redefinition of .subckt sky130_fd_pr__npn_11v0_w1p00l1p00, ignored
Warning: redefinition of .subckt sky130_fd_pr__pfet_g5v0d16v0, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_iso_pw, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_mim_m3_1, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_mim_m3_2, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po_0p35, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po_0p69, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po_1p41, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po_2p85, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_high_po_5p73, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po_0p35, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po_0p69, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po_1p41, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po_2p85, ignored
Warning: redefinition of .subckt sky130_fd_pr__res_xhigh_po_5p73, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_08p6x07p8_l1m1m2_noshield_o1, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_noshield_o1nhv, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_noshield_o1phv, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_l1m1m2m3m4_shieldm5, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_l1m1m2m3m4_shieldpom5, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_m1m4_noshield, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldm5, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_08p6x07p8_m1m2m3_shieldl1m5_floatm4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_04p4x04p6_m1m2m3_shieldl1m5_floatm4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1m5_floatm4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p3x11p8_l1m1m2m3m4_shieldm5_nhv, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p3x11p8_l1m1m2m3m4_shieldm5_nhv__base, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_l1m1m2m3m4_shieldpom5_x, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_02p9x06p1_m1m2m3m4_shieldl1_fingercap2, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_02p7x11p1_m1m2m3m4_shieldl1_fingercap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_02p7x21p1_m1m2m3m4_shieldl1_fingercap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_02p7x41p1_m1m2m3m4_shieldl1_fingercap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_02p7x06p1_m1m2m3m4_shieldl1_fingercap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p3x11p3_m1m2m3m4_shieldl1_wafflecap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_05p9x05p9_m1m2m3m4_shieldl1_wafflecap, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_44p7x23p1_pol1m1m2m3m4m5_noshield, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_08p6x07p8_m1m2m3_shieldl1, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_04p4x04p6_m1m2m3_shieldl1, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3_shieldl1, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_l1m1m2m3_shieldm4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_11p5x11p7_l1m1m2m3_shieldpom4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_06p8x06p1_l1m1m2m3_shieldpom4, ignored
Warning: redefinition of .subckt sky130_fd_pr__cap_vpp_06p8x06p1_m1m2m3_shieldl1m4, ignored

Circuit: ** sch_path: /home/ota/workspace/ifpb_enge_cia/praticas e projetos/01 transistor/testetransistor.sch

option SCALE: Scale is set to 1e-06 for instance and model parameters
option SCALE: Scale is set to 1e-06 for instance and model parameters
Netlist line no. 5:
 Mismatch: 14 formal but 1 actual params.
mult=1;l=$;w=$;nf=$;ad=$;as=$;pd=$;ps=$;sa=$;sb=$;sd=$;nrd=$;nrs=$;swx_nrds=$;
Netlist line no. 5:
Syntax error: letter [$]
Netlist line no. 5:
 Expression err: $
Netlist line no. 5:
 Formula() error.
      |mult=1; l=$; w=$; nf=$; ad=$; as=$; pd=$; ps=$; sa=$; sb=$; sd=$; nrd=$; 
Netlist line no. 803:
Undefined parameter [l]
Netlist line no. 803:
Cannot compute substitute
Netlist line no. 837:
Undefined parameter [swx_nrds]
Netlist line no. 837:
Cannot compute substitute
Netlist line no. 4617:
Undefined parameter [m]
Netlist line no. 4617:
Cannot compute substitute
Netlist line no. 25:
Undefined parameter [m]
Netlist line no. 25:
Cannot compute substitute
Netlist line no. 27:
Undefined parameter [m]
Netlist line no. 27:
Cannot compute substitute
Numparam expansion errors: Run Spice anyway? y/n ?
```

when its commented simulations finishes without errors.

RTimothyEdwards, 2025-05-15:
Something is clearly very wrong here, but I'm not able to reproduce anything like it on my end.

The "combined" models is a combination of the "continuous (-binned)" models and the "discrete (-binned)" models.  The "discrete" models are what's left over from the original models after removing all of the devices that have continuous-binned models.  The "sky130.lib.spice" file loads the continuous-binned models first, followed by the remaining discrete-binned models.

In your simulation, it appears that where the file "corners/tt.spice" has included "all.spice", it has read *all* of the original models, which is causing the warnings.  I don't understand the error that caused the simulation to halt, but the warning messages already indicate that something is very wrong.

"corners/tt.spice" should include eight other model files, most of them for special/unusual devices, and one of them for "all.spice".  It *should* read "all.spice" from the same directory (corners/).  "all.spice" (the version in combined/corners) is 31 lines long and  includes only devices like ESD diodes, varactors, SONOS and SRAM devices.

Probably the best way to figure out what is going wrong is to introduce errors into "all.spice" and see if those are being caught by ngspice, or if ngspice is reading the wrong file.

If you are not using any of the "special" (ESD, RF, SONOS, SRAM, etc.) devices, then your approach works fine.  I would like to know what has gone wrong with the open_pdks installation, though.

OtacilioNeto, 2025-05-16:
Hi, I made this script to set up Magic/Xschem/Ngspice/SkyWater PDKs on Ubuntu 24.04. You can reproduce the issue by running the script and simulating the SPICE netlist afterward.

```
#!/bin/sh

sudo apt update
sudo apt upgrade -y

sudo apt install wget xterm mesa-utils libgl1-mesa-dri libegl-mesa0 libglx-mesa0\
	build-essential autoconf automake make libx11-dev libxrender-dev \
	libx11-xcb-dev libcairo2-dev tcl8.6-dev tk8.6-dev bison libxpm-dev \
	libjpeg-dev flex git libxaw7 libxaw7-dev libreadline8 libreadline-dev -y

export XSCHEM=3.4.7
export MAGIC=magic-8.3.526
export NGSPICE=ngspice-44.2
export OPEN_PDKS=1.0.518
export PDK_DIR=PDKs

export DIR="$HOME/EDA"
export MAGIC_HOME=${DIR}/magic
export XSCHEM_HOME=${DIR}/xschem

export PATH=${MAGIC_HOME}/bin:$PATH
export PATH=${XSCHEM_HOME}/bin:$PATH
export PATH=${DIR}/ngspice/bin:$PATH

export PDK_PATH=${DIR}/${PDK_DIR}
export nproc=1 # para poder executar em máquinas com pouca memória como as virtualizadas

if [ ! -d ${XSCHEM_HOME}/bin ]; then 
	if [ ! -f ${XSCHEM}.tar.gz ]; then
		wget https://github.com/StefanSchippers/xschem/archive/refs/tags/${XSCHEM}.tar.gz
	fi
	tar -zxvf "$XSCHEM.tar.gz"
	cd xschem-${XSCHEM}
	./configure --prefix=$HOME/EDA/xschem
	make && make install
	cd ..
fi

if [ ! -d ${MAGIC_HOME}/bin ]; then 
	if [ ! -f "$MAGIC.tgz" ]; then
		wget "http://opencircuitdesign.com/magic/archive/$MAGIC.tgz"
	fi
	tar -zxvf "$MAGIC.tgz"
	cd $MAGIC
	./configure --prefix=$HOME/EDA/magic --enable-cairo-offscreen
	make && make install
	cd ..
fi

if [ ! -d ${DIR}/ngspice/bin ]; then
	if [ ! -f "$NGSPICE.tar.gz" ]; then 
		wget https://sourceforge.net/projects/ngspice/files/ng-spice-rework/44.2/ngspice-44.2.tar.gz
	fi
	tar -zxvf "$NGSPICE.tar.gz"
	cd $NGSPICE
	./configure --prefix=$HOME/EDA/ngspice --with-x --enable-openmp --enable-cider
	make && make install
	cd ..
fi

if [ ! -d ${PDK_PATH}/sky130B ]; then 
	if [ ! -f "$OPEN_PDKS.tar.gz" ]; then
		wget "https://github.com/RTimothyEdwards/open_pdks/archive/refs/tags/$OPEN_PDKS.tar.gz"
	fi
	tar -zxvf "$OPEN_PDKS".tar.gz
	cd "open_pdks-$OPEN_PDKS"
	export SHARED_PDKS_PATH=${PDK_PATH}
	./configure --prefix=${PDK_PATH} --enable-sky130-pdk --enable-sram-sky130 --enable-sram-space-sky130
	make && make install && make veryclean
	cd ..
fi

cat > ${DIR}/env.sh << 'EOF'
export PDK_ROOT="${HOME}/EDA/PDKs"
export PATH=$PATH:$HOME/EDA/magic/bin:$HOME/EDA/ngspice/bin:$HOME/EDA/xschem/bin
alias magic="${HOME}/EDA/magic/bin/magic -rcfile ${HOME}/EDA/PDKs/sky130B/libs.tech/magic/sky130B.magicrc"
alias xschem="${HOME}/EDA/xschem/bin/xschem --rcfile ${HOME}/EDA/PDKs/sky130B/libs.tech/xschem/xschemrc"
alias ngspice="${HOME}/EDA/ngspice/bin/ngspice"
EOF

```

and then 
`source $HOME/EDA/env.sh`